monorkin/omgkbd #9: Choose housing materials

Xenosb, 2017-10-22:


stankec, 2017-10-23:
Hm... Perhaps we should design all of them, just so that anybody that stumbles upon this repo can pick his own favorite.
* 3D printing ABS/PLA isn't that hard to do or design
* I guess that most of the work for 3D printing can be reused for CNC-ed wood
* Cut metal and acrylic are a ??? for me :)

Xenosb, 2017-10-23:
I'd suggest that we start with layered design.

/''''\ /''''\ /''''\ /''''\ /''''\ /''''\ /''''\ /''''\ /''''\ /''''\ /''''\      Keycaps
_______________________________      Acrylic 3mm 
..............................................................      PCB
==============================     Acrylic 6mm

stankec, 2017-10-23:
That seems like the fastest way to do it. Though if we are building a 60% layout then the top level acrylic is unnecessary, since the PCB won't be visible from the keys.

Xenosb, 2017-10-23:
Top layer would provide rigidity and protection to the electronics. eg.
metal shards can't scratch or damage the PCB if there is a layer on top of
it.
As alternative, we can have an aluminium top which would be thinner and
look more premium.

stankec, 2017-10-23:
Aha. I thought that the PCB would have a solder mask. and that it would protect the traces from dust and debris. My current keyboard has no top layer either, and if I remember correctly @shiranaii  also noted that she preferred a naked PCB.

Though this depends if we are going for a full-sized keyboard. In which case we need a top cover.

A 75% keyboard.
[image]

Or a 60% keyboard.
[image]

In the latter two cases, a top plate isn't required and IMO should be omitted.

Xenosb, 2017-10-23:
In that case, let's do it without a cover.

So, this means that we only need to cut some material for the bottom plate
and drill some holes for screws holding the PCB in place?
This is great!

shiranaii, 2017-10-23:
i agree, no need for the upper acrylic. a bare PCB is just fine for a 60%, considering the keys cover the whole upper surface (presuming we are going with a standard 60% layout without the arrow keys)

closed 2017-10-24 by Xenosb