monorkin/omgkbd #12: 3D print a test keycap

Xenosb, 2017-10-23:
Try manufacturing a keycap with good enough ergonomics

stankec, 2017-10-23:
This is my first attempt. I'll try to lower the layer height and try a few different things.
[image]
[image]

stankec, 2017-10-23:
I think we have a winner.
The one on the bottom is printed with 0.6 layer height and 105% flowrate.
[image]

Xenosb, 2017-10-23:
So are we going to order keycaps or 3d print them?

stankec, 2017-10-23:
I'll buy the ones mentioned in #10 and I'll produce a set of 3D printed ones then we can check if it works as expected.
I'd like for this repo to have everything to build your own keyboard - that includes keycaps, body, ...

Xenosb, 2017-10-23:
Then I'll close this issue and add print keycaps and buy keycaps tasks.

closed 2017-10-23 by Xenosb