monorkin/omgkbd #22: Decide on a final product design

Xenosb, 2017-10-25:


stankec, 2017-10-25:
COuld you explain what needs to be specified. Or should we write down everything?

Xenosb, 2017-10-25:
We need to define all of the parts - housing, PCB, connectors and other features. This will also be an outcome of the Sunday meeting.

shiranaii, 2017-10-29:
Conclusion (meeting):
* 60% layout
* ANSI layout
* LEDs -  [RGBs](https://www.aliexpress.com/item/10-1000pcs-WS2812B-4pins-5050-SMD-Black-White-version-WS2812-Individually-Addressable-Digital-RGB-LED-Chip/32453497583.html?spm=2114.search0104.3.26.CUHZWm&ws_ab_test=searchweb0_0,searchweb201602_4_10152_10065_5700015_10151_10344_10068_10345_10342_10343_10340_10341_5760015_10541_10084_10083_10307_10301_5640015_10303_10060_10155_10179_10154_10056_10055_5750015_10539_10538_5370015_10537_10312_10536_10059_10313_10314_10534_10533_100031_9912_10103_10073_10102_10142_10107-9912,searchweb201603_30,ppcSwitch_5&btsid=374f787a-0ea8-4bd9-8a32-4ee08bb32883&algo_expid=6b87b88c-a782-4c93-9bc5-a38d42409d17-3&algo_pvid=6b87b88c-a782-4c93-9bc5-a38d42409d17) under the rim and switches 
* USB C connector on the left side
* clear housing case for the keyboard
* discussed painting the top side of the PCB
* create .DWG for the top mechanical layoutb